wavedrom/wavedrom #247: Fragile usage of For...In statement

acotiuga, 2019-08-15:
WaveDrom uses the `For...In` statement (https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/for...in ) in a couple of places to iterate over arrays and objects without checking `hasOwnProperty()`, which is very fragile and error-prone because: 
* The `For...In` loop statement will return all enumerable properties, including those with non–integer names **and those that are _inherited_**.

Any JavaScript code that extends the Array or Object prototype will break WaveDrom. In my particular case, the Array and Object prototypes are extended by Prototype.js

drom, 2019-08-15:
Good point. I don't like `for in` anymore. How about we change them to `.map()`?

drom, 2019-08-15:
Will this style work for your case?
https://github.com/wavedrom/wavedrom/commit/3fc3c8fd78156afa4b21bace7c33f67e2a5c7c13

acotiuga, 2020-09-13:
Yes, it works. Thanks!

closed 2020-09-13 by acotiuga