wavedrom/wavedrom #253: SVG image not parsed/created inside jsonmlParse

gabrielrad, 2019-09-02:
Inside the jsonmlParse function, innerHTML it's used to parse the content returned by onmlStringify and create a svg image child:
```
function jsonmlParse (arr) {
    var el = document.createElementNS(w3.svg, 'g');
    el.innerHTML = onmlStringify(arr);
    return el.childNodes[0];
}
```
My user agent is _Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/538.15 (KHTML, like Gecko) Version/8.0 Safari/538.15_ and it does not parse the content assigned to innerHTML.

I solved this problem by doing the following:

```
function jsonmlParse (arr) {
    arr[1].xmlns = "http://www.w3.org/2000/svg";
    arr[1]['xmlns:xlink'] = "http://www.w3.org/1999/xlink";
    var svgStringify = onmlStringify(arr).replace(/&/g, "&amp;");
    var doc = new DOMParser().parseFromString(svgStringify, "image/svg+xml");
    return doc.firstChild;
}
```

Can this make it in a feature release? Thanks.

closed 2019-11-28 by drom

drom, 2019-11-28:
Please try new version.

gabrielrad, 2019-12-10:
Thank you very much!

gabrielrad, 2019-12-10:
@drom Can I ask you when it will be the next official release?

commit 90413c07 (2019-11-28): fixes #253 ; create DOM node from JSONML