wavedrom/wavedrom #254: Uncaught TypeError: e.charCodeAt is not a function

mithro, 2019-09-02:
On the latest chrome, when using [the editor](https://wavedrom.com/editor.html) with the following JSON;

```json
{
  "reg_definition": {
      "reg_name": "DATA",
      "reg_description": "Data from the last SETUP packet.  It will be 10 bytes long, because it will include the CRC16.  This is a FIFO; use `DATA_ACK` to advance the queue.",
      "reg": [
          { "name": "DATA",   "bits": 8, "attr": "RO", "description": "The next byte of SETUP data" }
      ]
  }
}
```

I get the following error in the Chrome console;
```
Uncaught TypeError: e.charCodeAt is not a function
    at t.exports (wavedrom.min.js:3)
    at t.exports (wavedrom.min.js:3)
    at t.exports (wavedrom.min.js:3)
    at t.exports (wavedrom.min.js:3)
    at t.exports (wavedrom.min.js:3)
    at t.exports (wavedrom.min.js:3)
    at Object.t.exports [as EditorRefresh] (wavedrom.min.js:3)
    at c (init.js:129)
```

drom, 2019-09-03:
It should more like:
```json
{
  "reg": [
    {"name": "DATA", "bits": 8, "attr": "RO", "description": "The next byte of SETUP data" }
  ]
}
```

mithro, 2019-09-03:
@drom Any chance the editor could report that you have invalid JSON / it doesn't understand the JSON output?

drom, 2019-09-03:
That we can do.

xobs, 2020-03-16:
I just ran into this issue after upgrading wavedrom -- the older version worked, but the newer version silently failed with only that error in the console -- `TypeError: e.charCodeAt is not a function`

Could there be a more descriptive error generated when JSON isn't valid?  In this case, there were trailing commas, which wavedrom apparently used to accept but now no longer does.