wavedrom/wavedrom #256: request for color option for the complete signal wave

dkgoyal, 2019-09-20:
thank you very much for creating this awesome tool for waveform generation. 

It will be very helpful if we can color a signal waveform with some option . 
That feature will be very useful.

drom, 2019-10-14:
@dkgoyal are you thinking about line, background, fill or text colors?
Could you provide some examples that would be useful?

grigoraghanyan, 2019-11-21:
I agree with enhancement request. We are using wavedrom as one of options to visualize waveforms in our reports and there is need to visually (by color)  mark some signals (both bus and bit) and also have waveforms colored with different color (red is preferable) starting from the frame (or only at the frame) where FAIL happens. Also please make the available colors more visually "non-neutral". It would be better to have exact Red, Green and Yellow colors instead of supported ones.

drom, 2019-11-21:
@grigoraghanyan the color scheme can be changed using skin mechanism described here: https://github.com/wavedrom/wavedrom/blob/master/unpacked/README.md
For other ways of coloring individual bits of a diagram, I would need some case by case visual examples to proceed.

raulbehl, 2020-12-11:
I think #271 implements the needed support for this? 

@drom : Here is what gtkwave allows you to do and I believe this is what was requested here: 
[image]