wavedrom/wavedrom #259: Waveforms with 26+ Unmarked Signal Nodes

Ganesh-AT, 2019-09-29:
Windows 10 / Firefox

I am trying to create WaveDrom waveforms similar to Figures 3, 4, and 5 of the datasheet here:

https://www.mrcy.com/siteassets/product-datasheets/rfm/eeprom-wme128k8-xxx.pdf

I have some basic idea of JavaScript, and can superficially understand the 'brick' architecture used by WaveDrom to generate waveforms.

Since these waveforms have a lot of edges / markings, I am running into a number of issues as below:

(1) The 'signal..node' notation allows for up to 26 unmarked nodes with [A-Z] : Is there a way to accommodate more than 26? i.e, instead of just a single upper-case character, have a way to use an upper-case string for a single node?

(Suggestion: each node can be a single character or a string enocldes with delimiters  - say, '-' or ':')

I also have a few other questions related to the above waveforms:

(1) Is there a way to alter the size of the font used in the edge markings? Similarly, a way to alter the size of the signal names? Also, is it possible to denote a '#' signal like WE# or CS# in the above PDF with WE or CS and a line on top?

(2) I looked at the currently available bricks in the different skins, and I do not see anything that can help with the sort of edges shown in the the waveforms in the PDF - mainly, have each rising or falling edge divided into three (or four) levels, and the ability to give node names to these levels. Is there a way to currently create those types of waveforms with WaveDrom? Just want to make sure I am not missing anything obvious in the feature list..

(3) I would like to create waveforms for signals with pre-defined setup and hold times - currently, I am using the phase feature, but 0.5 seems to be the only reliably supported non-integer value. I have 3 or 4 signals that each have different setup and hold times. Is there a way to achieve the requirement? This is a proof-of-concept I am trying to create : https://bit.ly/2nJel9O , but I fear the period / phase combinations I am creating do not make immediate sense compared to what could be achieved with a setup / hold time specification - where setup and hold times could be specified as decimal numbers less than 1.

drom, 2019-09-29:
> (1) The 'signal..node' notation allows for up to 26 unmarked nodes with [A-Z] : Is there a way to accommodate more than 26? i.e, instead of just a single upper-case character, have a way to use an upper-case string for a single node?

You can use escaped sequences like `\xF7` or `\u1234` Unicode have a lot of them

drom, 2019-09-29:
> (1) Is there a way to alter the size of the font used in the edge markings? Similarly, a way to alter the size of the signal names? Also, is it possible to denote a '#' signal like WE# or CS# in the above PDF with WE or CS and a line on top?

You can use `<o>WE` tag or any other https://github.com/drom/tspan markup tricks

Ganesh-AT, 2019-09-30:
Thanks for the clarifications. Other than the font-size issue and the multi-level rising edge, this question thread can be closed. For the font size issue, I see tspan could potentially help (I see the font-size being modified for the subscripts case in parse.js), but I imagine it could lead to potential rendering side-effects. 

Do you suggest that I close this thread and open a separate one for the multi-level rising and falling edges? Another example that I came across, which might involve the need for the same is Figure 29 in this PDF : http://datakey.com/downloads/223-0017-006_REVF_SPIFlashInterfaceSpec_SBMb.pdf : particularly, the annotations for tSRF and tSRT. Do you have some tricks that can be used with the current infrastructure to render those?

drom, 2019-10-07:
Yes, please open a separate issue on multi-level rising and falling edges.

closed 2019-10-07 by drom