wavedrom/wavedrom #264: Bus Transitions to High-Impedance State Not Rendered Correctly

Ganesh-AT, 2019-10-18:
WaveDrom Online Editor on Firefox 70.0.b3 / Windows 10 x64
WaveDrom Online Editor on Firefox 60.8.0esr / RHEL 6 

While attempting to create a signal with transitions from valid bus state to high-impedance, and it appears that the transition is not rendered correctly

https://tinyurl.com/y2lfksxf

[image]

Expected behavior - The change from multi-bit bus value to z must involve sharp lines (similar to the z to multi-bit bus value change) and not curved ones

Observed behavior - We get curved lines while transitioning from multi-bit to z, as shown in the picture above.

drom, 2019-10-18:
It is intended and correct behavior. The drawing represents the fact that transition into High Impedance state takes time.

Ganesh-AT, 2019-10-18:
Thank you for the clarification.

closed 2019-10-18 by Ganesh-AT

derchr, 2025-08-05:
Is there a way to disable this?

drom, 2025-08-11:
@derchr what is your use case?