wavedrom/wavedrom #266: how to place a edge line on clock edge

parafux, 2019-11-05:
I use this example:
{signal: [
  {name: 'clock', wave: 'p....',  
                  node: '.a', },
  {name: '', wave: ''},
  {name: '', wave: ''},
  {name: '', wave: '',
                  node: '.b', },
],
  edge: [
    'a-b',
  ]
}

How can i get this vertical line exactly at the clock edge?
How can i remove the a and b?

thank you
Andreas

parafux, 2019-11-05:
OK, i found a solution for my first question

{signal: [
  {name: '', wave: '',  
                  node: '.a',phase: 0.15 },
  {name: 'clock', wave: 'p....'},
  {name: '', wave: ''},
  {name: '', wave: ''},
  {name: '', wave: '',
                  node: '.b',phase: 0.15 },
],
  edge: [
    'a-b',
  ]
}

Totolitoto310, 2019-12-20:
For question 2:

{signal: [
{name: '', wave: '',
node: '.A',phase: 0.15 },
{name: 'clock', wave: 'p....'},
{name: '', wave: ''},
{name: '', wave: ''},
{name: '', wave: '',
node: '.B',phase: 0.15 },
],
edge: [
'A-B',
]
}