wavedrom/wavedrom #272: [svg.wavedrom.com] Using the JSON in the link with Base64 encoding.

alexisfrjp, 2019-12-10:
I would like to use use the svg server without having a different file for Gitlab/Github that doesn't support the preview.

**In the readme:**
```
![signal step4](https://svg.wavedrom.com/{signal:[{name:'clk',wave:'p......'},{name:'bus',wave:'x.34.5x',data:'head body tail'},{name:'wire',wave:'0.1..0.'}]})
```
That doesn't work, I guess it's due to spaces and other special characters.

It would be nice to add the support using BASE64.

**Example:**
```
{ signal: [
  { name: "clk",         wave: "p.....|..." },
  { name: "Data",        wave: "x.345x|=.x", data: ["head", "body", "tail", "data"] },
  { name: "Request",     wave: "0.1..0|1.0" },
  {},
  { name: "Acknowledge", wave: "1.....|01." }
]}
```
**Base64's equivalent:**
```
eyBzaWduYWw6IFsKICB7IG5hbWU6ICJjbGsiLCAgICAgICAgIHdhdmU6ICJwLi4uLi58Li4uIiB9LAo
gIHsgbmFtZTogIkRhdGEiLCAgICAgICAgd2F2ZTogInguMzQ1eHw9LngiLCBkYXRhOiBbImhlYWQi
LCAiYm9keSIsICJ0YWlsIiwgImRhdGEiXSB9LAogIHsgbmFtZTogIlJlcXVlc3QiLCAgICAgd2F2ZTogIj
AuMS4uMHwxLjAiIH0sCiAge30sCiAgeyBuYW1lOiAiQWNrbm93bGVkZ2UiLCB3YXZlOiAiMS4uLi4
ufDAxLiIgfQpdfQ==
```
**To be able to use:**
```
![example](https://svg.wavedrom.com/base64
/eyBzaWduYWw6IFsKICB7IG5hbWU6ICJjbGsiLCAgICAgICAgIHdhdmU6ICJwLi4uLi58Li4uIiB9LA
ogIHsgbmFtZTogIkRhdGEiLCAgICAgICAgd2F2ZTogInguMzQ1eHw9LngiLCBkYXRhOiBbImhlYW
QiLCAiYm9keSIsICJ0YWlsIiwgImRhdGEiXSB9LAogIHsgbmFtZTogIlJlcXVlc3QiLCAgICAgd2F2ZTo
gIjAuMS4uMHwxLjAiIH0sCiAge30sCiAgeyBuYW1lOiAiQWNrbm93bGVkZ2UiLCB3YXZlOiAiMS4uL
i4ufDAxLiIgfQpdfQ==)
```

It should be very easy to support that.

drom, 2019-12-12:
Here is the server: https://github.com/wavedrom/lambda

alexisfrjp, 2019-12-12:
Moved it to https://github.com/wavedrom/lambda/issues/5.

closed 2019-12-12 by alexisfrjp