wavedrom/wavedrom #273: Signal name can't have "<" in it

pwiecha, 2019-12-17:
I'm using `->` and `<-` to denote the origin/destination of signals in the module.
I've notice that placing a `<` breaks the online editor, it thinks it's a html tag?
```
{signal: [
  {name: 'module A <- sig B', wave: '10........'},
```

Gives:
```
This page contains the following errors:
error on line 1 at column 41012: StartTag: invalid element name
Below is a rendering of the page up to the first error.
```

connorjan, 2020-01-17:
@pwiecha Although not a fix to the bug, I suggest you just copy and paste a Unicode arrow in there as a workaround: [example](https://wavedrom.com/editor.html?%7Bsignal%3A%20%5B%0A%20%20%7Bname%3A%20%27module%20A%20%F0%9F%A1%84%20sig%20B%27%2C%20wave%3A%20%2710........%27%7D%0A%5D%7D%0A).

Wikipedia has a [list of arrows](https://en.wikipedia.org/wiki/Arrow_(symbol)) you can copy and paste from.

drom, 2020-04-20:
should be fixed in v2.4.2

pwiecha, 2020-04-20:
Tested it using the online editor (I assume it's running newest version), it doesn't break anymore. Thanks @drom!

closed 2020-04-20 by pwiecha