wavedrom/wavedrom #288: Is it possible to use Horizon Chart as a row? [FeatureRequest]

OlegJakushkin, 2020-03-27:
[Horizon Chart (with source code)](https://observablehq.com/@d3/realtime-horizon-chart) seems like a perfect addition to wavedrom capable to represent math functions and arbitrary data values. I wonder is it possible and how one can add a custom row renderer to wavedrom?

So what I would like to see
[image]

And in terms of JSON, it would be something like
```json
{ signal: [
  { name: "func", wave: 'f.f(t){/*js code returning a float value*/ return Math.random() * t }' },
  { name: "vals", wave: 'v.x.345x', data: ["90", "300", "120"]  },
]}
```
But I focus primarily on heaving at least JS accessible renderer...
So is such thing possible and how one would do it?

drom, 2020-03-29:
Well, at this point, WaveJson is just a data structure. And different tools, written in different languages JS, Python, Ruby, etc. can consume or produce it. Adding JS callback would definitely be a hostile move. Also, I am trying to imagine how "Horizon Chart" would be useful to Chip-Designer audience? There is a long-standing request on having "Analog" waves #189  #68 that might be related?

OlegJakushkin, 2020-03-30:
Yes, it is in general alike what I want, rendering approaches do not matter.