wavedrom/wavedrom #291: Edge labels can't have "<" in it

CacahueteNC, 2020-04-19:
Hi there, and thanks for the great software. This issue is strange as I already generated tons of waveforms with a "<" in an  edge label...

[image]
It was with an old version that I first saw it so I switched to the latest and it still has it.

[Here is an example](https://wavedrom.com/editor.html?%7B%20signal%3A%20%5B%0A%20%20%7B%20name%3A%20%27Reset%27%2C%0A%20%20%20%20%20wave%3A%20%20%2701%7C.......%7C01u%27%2C%20%0A%20%20%20node%3A%20%27...........E...%27%20%7D%2C%0A%20%20%7B%7D%2C%0A%20%20%7B%20name%3A%20%27Input%27%2C%0A%20%20%20%20%20%20wave%3A%20%27x0%7C.1x....%7C.0d%27%2C%20%0A%20%20%20%20%20%20node%3A%20%27....A.F.......%27%20%7D%2C%0A%20%20%7B%7D%2C%0A%20%20%7B%20name%3A%20%27Output%27%2C%0A%20%20%20%20%20%20wave%3A%20%270.%7C......1%7C0.d%27%2C%0A%20%20%20%20%20%20node%3A%20%27.........C.F..%27%20%7D%2C%0A%20%20%0A%20%20%7B%20%20%20node%3A%20%27....B....D.....%27%20%7D%2C%0A%20%20%5D%2C%0A%20%20edge%3A%20%5B%0A%20%20%20%20%27A-B%27%2C%20%27C-D%27%2C%20%27A~%3EC%27%2C%20%27B%3C-%3ED%20T%20%3C%2045s%27%2C%20%27E-%3EF%27%0A%20%20%5D%0A%7D) with the online editor silently crashing. The desktop editor returns the following error: `error on line 1 at column 47283: StartTag: invalid element name`

Is there a way to have an edge label containing the "<" character? Might be related to #273.

Thanks for the help!

drom, 2020-04-20:
should be fixed in v2.4.2

CacahueteNC, 2020-04-22:
Indeed, it works in the web editor now, this was pretty quick! Thanks a lot @drom 😎
I'll try it at work tomorrow 👌

closed 2020-04-22 by drom