wavedrom/wavedrom #293: Content Security Policy violation

smariel, 2020-04-21:
I wanted to use WaveDrom in an [Electron](electronjs.org/) desktop application. But it does not comply with strict (`default-src 'self'`) [Content Security Policy](https://developer.mozilla.org/en-US/docs/Web/HTTP/CSP), which is essential in modern web app.

First, as mentioned in #75, eval() is used and is a critical vulnerability. Using the `unsafe-eval` policy enables it but it's not safe at all.

Then, style is injected multiple times. This can be authorized by using the `unsafe-inline` policy. But is not safe either.
https://github.com/wavedrom/wavedrom/blob/a9a077a617e1596104bd50d48ecb7019319a8a0e/lib/render-wave-element.js#L13 https://github.com/wavedrom/wavedrom/blob/a9a077a617e1596104bd50d48ecb7019319a8a0e/lib/process-all.js#L41

WaveDrom would perfectly fit my app. But I can't use it with those vulnerabilities. Especially the eval function. Is this something you could work on ? The only other solution I found is to use [chartjs](https://github.com/chartjs/Chart.js), which is already in my dependencies, to reproduce a diagram like WaveDrom. But this wont be as good.
By the way, if it can help, they also had [the same issue](https://github.com/chartjs/Chart.js/issues/5208) and managed to fix it.

drom, 2020-04-21:
@smariel there are five phases inside `wavedrom` module:
1. Convert source `string` into source `object`. The source can be JSON, JSON5 or JS code that takes parameters and returns 'object'. Here we use `eval()`. It can be JSON5 parser, but it would limit current use cases.
2. Validate source `object` -- (safe)
3. Convert source `object` into JSONML `object` -- (safe) `lib/render-any.js`
4. Convert JSONML `pbject` into SVG `string` -- (safe) `onml/stringify'
5. Insert SVG `string` into DOM of the browser. That is where we are doing `.insertBefore` , `.innerHTML` etc.

Phases 2, 3, and 4 are completely safe and can be required and used separately from the rest of the code.

smariel, 2020-04-21:
Thank you very much for those explanations. And the example is perfect.

Just a suggestion, for the first phase, would it be possible to specify the method to use with an option object ?

```
WaveDrom.ProcessAll({
  source: 'json' // or 'js', as default option
});
```

or maybe
```
WaveDrom.ProcessAll({
  method: 'parse' // or 'eval', as default option
});
```

And thank you for your work on WaveDrom. This a is a fantastic software.

drom, 2020-04-21:
Good idea. As the minimum level of source input I would use https://github.com/json5/json5

smariel, 2020-04-23:
I spent some time working with the SVG string returned by `wavedrom.onml.stringify(wavedrom.renderAny(...));` to make my own Phase 5.

After using the [DOM Parser](https://developer.mozilla.org/en-US/docs/Web/API/DOMParser) to analyse the SVG, which I'm absolutely not familiar with, it's more clear now. The problem is not `.insertBefore` or `.innerHTML` themselves but the content of the SVG. It contains an embedded `<style>` which is not allowed by strict CSP.
One way to correct that, would be to only embed classes and generate a `.css` file for each skin. This could be done by done by modifying `svg2js.js` to output a css file also. Maybe with a specific attribute.
https://github.com/wavedrom/wavedrom/blob/a9a077a617e1596104bd50d48ecb7019319a8a0e/bin/svg2js.js#L20-L21
To avoid breaking changes, this could also be specified within an option object.
```javascript
WaveDrom.ProcessAll({
  source: 'json5', // js as default
  embedStyle: false // true as default
});
```

For now, the solution I choose is to use [Fabric](https://github.com/fabricjs/fabric.js), which is already loaded in my app, to parse the SVG and recreates canvas objects by itself.