wavedrom/wavedrom #294: Schematic broken in 2.4.2

CacahueteNC, 2020-04-23:
Hi, I'm running the latest 2.4.2 desktop version on Win 10 (64) and it seems that nested schematics (more than one stage of logic gates) is now broken. 

The issue can be observed in the [tutorial section](https://wavedrom.com/tutorial2.html) of the website or in the [online editor](https://wavedrom.com/editor.html?%7B%20assign%3A%5B%0A%20%20%5B%22out%22%2C%0A%20%20%20%20%5B%22%7C%22%2C%0A%20%20%20%20%20%20%5B%22%26%22%2C%20%5B%22~%22%2C%20%22a%22%5D%2C%20%22b%22%5D%2C%0A%20%20%20%20%20%20%5B%22%26%22%2C%20%5B%22~%22%2C%20%22b%22%5D%2C%20%22a%22%5D%0A%20%20%20%20%5D%0A%20%20%5D%0A%5D%7D). [Simple gates](https://wavedrom.com/editor.html?%7B%20assign%3A%5B%0A%20%20%5B%22g0%22%2C%20%5B%22%5E%22%2C%20%22b0%22%2C%20%22b1%22%5D%5D%2C%0A%20%20%5B%22g1%22%2C%20%5B%22%5E%22%2C%20%22b1%22%2C%20%22b2%22%5D%5D%2C%0A%20%20%5B%22g2%22%2C%20%5B%22%5E%22%2C%20%22b2%22%2C%20%22b3%22%5D%5D%2C%0A%20%20%5B%22g3%22%2C%20%5B%22%3D%22%2C%20%22b3%22%5D%5D%0A%5D%7D) do work.

closed 2020-04-23 by drom

drom, 2020-04-23:
Should be fixed in v2.4.4

CacahueteNC, 2020-04-23:
You rock soooo much @drom 🎸 !! It is okay in the browser, as far as my sample goes 😃

commit 147bb2d6 (2020-04-23): fixes #294 ; Schematic broken in 2.4.2 ; escaped symbols in text and title with tspan