wavedrom/wavedrom #298: correctly using phase

nachiket, 2020-05-19:
I'm not sure I fully understand phase option in the drawing tool. Sorry, if this is covered in a document somewhere, but I'm only operating off the tutorial.

Here's an example where a phase shifts the entire waveform to the right. Can I have it such that the spillover is clipped, and the initial gap isn't present i.e. same time range as rest of the plot? I'm trying to show delta cycles close to the transition, but cannot figure out a way to position the `c` signal without gap and spillover effects.

[image]

drom, 2020-05-20:
I agree that is it ugly.
The gap on the left is because nothing defined to the left of the first 0 in your `wave`.
The waveform is constructed from bricks. Brick is the minimal quantum of drawing.
Every period = 2 bricks. It should be no spills if your phase is an integer number.

nachiket, 2020-05-20:
Hmm, I think I understand, the number of bricks I'd need is the max number of delta cycles I want in a clock period or time period.

drom, 2020-05-20:
yes, if you use `config:{skin:'narrow'}` and
`period: 2` you can use phase granularity 0.5
`period: 4` you can use phase granularity 0.25