wavedrom/wavedrom #299: Regression: tspan in signal name doesn't work anymore

frasten, 2020-06-04:
I'm trying to change the label color for a waveform. It used to work in some previous version by putting a `tspan` with a custom style inside the `name` of the waveform.

Unfortunately this code isn't working anymore (I'm using the official wavedrom editor):
```
{signal: [
    {name: ['tspan', {style: 'fill: red'}, 'Test'], wave: '0HL'}
]}
```
I get this error in the browser console::
> TypeError: str.charCodeAt is not a function

For comparison, the code without the styling is as simple as:
```
{signal: [
    {name: 'Test', wave: '0HL'}
]}
```

Thank you for your awesome tool.

drom, 2020-06-07:
You can use `tspan` styling in most of the text labels, except the `name`.
The `name` is special because I have to measure the width of this text to let enough space on the left.
You can use some https://github.com/drom/tspan markup inside the text.

frasten, 2020-06-17:
Thank you for your reply. Is there any other way to change the label color? I could have misread the docs, but tspan maskup doesn't seem to be able to change it.

jwhavican, 2020-08-19:
I agree that it would be really useful to be able to change the label color.

By the way, thanks for developing this tool, it is awesome!

commit 2982271d (2022-03-07): don't use textWidth if signal name is a tspan

closes #299

We were using `getBBox().width` before "browserless version" but we
cannot do this anymore with tspans. Therefore, we skip measuring
for this lane. This will only work if there are other lanes that are
wider and have a normal name. For cases when this doesn't work, the
user must explicitely set textWidth on the tspan.
closed 2022-03-07 by drom