wavedrom/wavedrom #302: place text on line

Johnlon, 2020-07-19:
Online editor.

I want to put text along a line in a similar way to how "data" works.
However, I want the text on high or low lines - I don't want the text shown inside a data lozenge.

I can approximate what I want using edges but the problem here is that I don't want to see the lines of the edges.
[image]

t-moe, 2020-08-15:
Yes. Please add labels for low/high lines (or any state?) similar to what "data" provides for "values".

I think this is really important and it was the first thing I was looking for.
Without it, I can't even label a simple RS232 transmission start bit.

drom, 2020-08-16:
Is the example below close to what you trying to achieve?
Empty extra lane explicitly introduced for labels:

```wavedrom
{signal: [
  {name: 'foo', wave: 'p....'},
  {                            node: '.A..B.'},
  {name: 'bar', wave: '01..0', node: '.a..b.'},
],
  edge: ['a-A', 'A-B some text', 'B-b']
}
```

[image]

t-moe, 2020-08-17:
I would suggest a simpler approach:
[image]

Maybe, you could add a 'alldata' property, to not break the functionality of the existing data property.
```
{signal: [ 
    {name: 'UART Example', 'wave': '104.......1u', 'alldata': ['','Start','data', 'ACK', '']},
]}

```

drom, 2020-08-17:
I understand now. Allowing `data` attribute on 0,1 and other segments do make sense.
It should not break anything. Will fix.

t-moe, 2020-08-17:
I'm having a hard time navigating through the source code, as large parts are undocumented. I couldn't pinpoint the location where you decide whether or not to render a label at a certain point in time.
I would really appreciate it if you could provide a fix.

Johnlon, 2020-08-17:
> I would suggest a simpler approach:
> [image]
> 
> Maybe, you could add a 'alldata' property, to not break the functionality of the existing data property.
> 
> ```
> {signal: [ 
>     {name: 'UART Example', 'wave': '104.......1u', 'alldata': ['','Start','data', 'ACK', '']},
> ]}
> ```

Yes that aligns with my need in my original post

ColePBryan, 2020-12-15:
@drom I have the code to match the above suggested implementation from @Johnlon . The one thing I am wondering is if this should be implemented in a backwards compatible way? It seems like this could break a lot of existing wavefiles when this is integrated. We could possibly make so it can be enabled in the config?

ColePBryan, 2020-12-15:
Actually I just saw the @Johnlon changed it from 'data' to 'alldata' which would be backwards compatible.

ColePBryan, 2020-12-15:
With the all data approach how should it be handled when both 'data' and 'alldata' are provided?

ColePBryan, 2020-12-15:
Here is an initial PR that would add the following functionality:

```json5
{
 signal: [ 
    {name: 'UART Example', 'wave': '104.......1u', 'data': ['','Start','data', 'ACK', '']}
    ],
 config: {'labelsOnOneAndZeroLines':true}
}
```

[image]