wavedrom/wavedrom #304: negative clock starting on low shows phantom edge

jeremyherbert, 2020-08-26:
Is it possible to have a negative clock with "N", but that starts on a low signal? For example, wavedrom shows this with "0...N......"

[image]

But what I am trying to show is this:

[image]

drom, 2020-09-13:
Agree, that is the bug.

Remillard, 2021-05-12:
Yep, I just ran into this as well.  SPI discontinuous clock with polarity 0 starts low and then has a rising edge mid bit period with no glitch.  I think it's going to be back to Visio for the time being, but hopefully this can be resolved.

drom, 2021-05-18:
How about this solution?

```
{signal: [
  {name: 'clk1', wave: 'l..N..'},
  {name: 'clk2', wave: 'l.hN..'},
  {name: 'clk3', wave: 'h..P..'},
  {name: 'clk4', wave: 'h.lP..'},
]}
```
[image]

`clk1` : Before the `Negative edge` you need to un-gate clock into `high` state.
`clk2` : that can be done with symbol `h` in the cycle right before the edge.

`clk3` : Before the `Positive edge` you need to un-gate clock into `low` state.
`clk4` : that can be done with symbol `l` in the cycle right before the edge.

Remillard, 2021-05-18:
I don't think that would work for me.  I needed this output:
[image]

`sclk` is low and then its first transition is low to high as the capturing edge in the middle of the bit period.  In none of the examples above would this be conveyed appropriately.  Hopefully at some point wavedrom will transition from 0 into a low to high clock without the glitch edge.

drom, 2021-05-18:
@Remillard will this work?

```
{signal: [
  {name: 'sclk',   wave: 'l.P.|.', phase: .5},
  {name: 'sdmosi', wave: '022<2|>2', data: 'WR A14 A13 A12'},
]}
```
[image]

Remillard, 2021-05-18:
Also is there better documentation than https://wavedrom.com/tutorial.html?  I did not recognize that `<` code.  Several years ago I was looking for something like that where it would control how wide a data phase followed by a discontinuity code.  Didn't know about it.

suzizecat, 2022-07-19:
Hi there, I got exactly on the same situation and the best I could get was this : 
```json
{
  "signal": [
    {"name": "\\symb{csn}",  "wave": "10.."                   },
    {"name": "\\symb{clk}",  "wave": ".<lh>NN"                },
    {"name": "\\symb{mosi}", "wave": "x222",  "data": "1 2 3" },
    {"name": "\\symb{miso}", "wave": "x222",  "data": "1 2 3" }
  ],
  "foot": {
    "tick": 0
  }
}
```
[image]

My take on this is that using `"0<.h>` should do the trick, but the dot in first position will not capture the `0`from outside the group and will default to `X`. Likewise, using the "group" (`<>` syntax) at the beggining of a wave will be treated as X. (otherwise `<0..h>L` would have done the trick.

I'll gladly report those issues, but I don't know the name of the "group" feature to lookup if it was already reported somewhere.