wavedrom/wavedrom #305: How do I represent time sensitive date accurately in wavedrom ?

DHARMPATEL29, 2020-09-11:
Please describe your system environment before submitting your Issue.
[image]

I attached two back to back transfer of the I2C protocol image. All mentioned times are random and just for reference. I want to represent the above image accurately in terms of timing in wavedrom. 

For example, if the Start condition is taking 3ns then the wave represents in wavedrom equal to 3ns.

Is there any way or inbuilt feature that helps me to represent timing accurately in wavedrom?

DHARMPATEL29, 2020-09-17:
Is is possible to draw timing diagram with actual timing scale ?
If possible than Can you please show me a way to draw timing diagram with accurate timing in wavedrom ?

drom, 2020-09-18:
WaveDrom diagram is constructed from building blocks of standard size. That limits your time step.
In your example, I see that everything is 0.5ns accurate. So one can draw it this way:

[image]

```js
{signal: [
  {node:'A.....B...................C..D.E...............F...G', period: 0.5},
  {name: 'SDA', wave: '1.0...3...3.|.3...4...5...1..0.6...6.|.6...7...10..1', data: 'A6 . A0 R/W ACK D7 . D0 ACK', period: 0.5},
  {name: 'SCL', wave: '1...0..1.0.1|0.1.0.1.0.1.0......1.0.1|0.1.0.1.0..1..', period: 0.5}
],
 config:{skin: 'narrow'},
 edge: ['A<->B 3ns', 'B<->C 20ns', 'C<->D 1.5ns', 'D<->E 1ns', 'E<->F 20ns', 'F<->G 2ns']
}
```