wavedrom/wavedrom #307: 'svg.wavedrom.com' doesn't seem to support config

K4zuki, 2020-09-15:
Please describe your system environment before submitting your Issue.

- MacOS 10.15.6
- Chrome 85.0.4183.83

---

- When input following code on GitHub comment, `config` doesn't give effect, i.e. single lane or default skin is applied.

```html
<img src="https://svg.wavedrom.com/{config:{bits: '32', lanes: '2'},reg: [{name: 'a', bits: 8},{name: 'b', bits: 8},{name: 'c', bits: 1,type:2}]}"/>
<img src="https://svg.wavedrom.com/{config:{skin:'narrow'},signal:[{name:'clk',wave:'p......'},{name:'bus',wave:'x.34.5x',data:'head body tail'},{name:'wire',wave:'0.1..0.'}]}"/>
```
[image]
[image]

drom, 2020-09-15:
it might be the issue with white space. Please try without it.

drom, 2020-09-15:
@K4zuki I have published new version, please try.