wavedrom/wavedrom #308: Wavedrom editor becomes unresponsive unusable with infinite loops

berkakinci, 2020-11-10:
Found on:
Windows 10 Pro x64
wavedrom-editor-v1.9.2-win-x64
wavedrom-editor-v2.4.2-win-x64
Reproduced online at:
https://wavedrom.com/editor.html

This is related to Issue #75 but a little more acute a problem.

I was editing a waveform returned from a function(){}().  While editing, I ended up with an infinite loop like:
`for(var i=0; i<10; i){};`
I never got a chance to edit the line to include `i++`.  Wavedrom editor hung and crashed.
Every time I open it, it reloads the same waveform definition object; hangs; crashes.
I can also replicate the same issue on the web-based editor.  In that case, I can break out of the behavior by clearing cache/cookies for this web site.
On the natively-installed (winx64) binary, I can't figure out where to recover my code and remove it to break out of this loop.

I propose that Wavedrom put a timeout on long-running functions to allow user to break out of such an issue.
In the short term: 1) where do I get the stored object from native editor; and 2) how do I clear it so I can run the native editor again?

drom, 2020-11-18:
yes, that is a bug. Will the timeout work when the main JS thread is blocked by an infinite loop? Or we better run this code in a web worker?

berkakinci, 2020-11-19:
I don't know the implementation details for the fix.  I just thought a (configurable?) timeout would be a good idea.
It's especially bad since the bad code is loaded on refresh (web) or re-launch (winx64) causing the whole install to be unusable.
In the short term, can you document the recovery procedure? I've found clearing stored web-site data for the web edition fixes it.  For the native install, I haven't been able to clear the problem.