wavedrom/wavedrom #320: Clocks and signals edges alignments

suzizecat, 2021-04-19:
Hello,
I was making a diagram when I saw that all signals except clocks seems to have a bit of phase offset.
This gives the feeling that the clock is *before* every data edge, as seen in this screenshot :

[image]

Here, it seems that the clock pulse comes *before* the write data change and the lock value change.
Even though it makes sense for the stored data to be after the clock, it doesn't for the two other signals.

I can phase -0.15 my clock to be in line with all signals but then all vertical "bars" indicating time seems misaligned. I could also move a bit the data signals but it then have the issue of having a gap at the begining or at the end...

Is there a proper way to fix this (our could this alignment be fixed ?)
This is what I would expect : 

[image]

My wavedrom code is : 
```text
{"signal":[
  {name:'clock', wave:"P.."},
  {name:'write data',wave:"x47", data:["data1","data2"]},
  {name:'lock',wave:"0.1"},
  {name:'stored value',wave:"x.7",data:["data2"]}
],
 config:{hscale:4}}
```

best regards,
Julien FAUCHER

zpqrtbnk, 2021-04-22:
Same question. Running with .15 phase everywhere, so there is a workaround, but...

drom, 2021-05-16:
This is intentional delay showing that any change of the signal happens after the clock edge.