wavedrom/wavedrom #322: [question]How can I draw a clk signal exactly like this example?

Huansion, 2021-04-26:
It's my first time to use wavedrom ,with edge browser.
Thanks for answering!!!!!!!
[image]

here is my code, it's not perfect:
{signal: [
  {name: 'clk1', wave: 'lN......L',phase:'0.5'},  
  {name: 'clk2', wave: 'hP......H',phase:'0.5'},
]}

[image]

Huansion, 2021-04-26:
I export a svg file and open it with Visio(2019) .But another question occurred, an annoying  extra blank  appearred  on the left which i can't remove it.

[image]

drom, 2021-05-06:
> It's my first time to use wavedrom ,with edge browser.
> Thanks for answering!!!!!!!
> [image]
> 
> here is my code, it's not perfect:
> {signal: [
> {name: 'clk1', wave: 'lN......L',phase:'0.5'},
> {name: 'clk2', wave: 'hP......H',phase:'0.5'},
> ]}
> 
> [image]

it does look like a bug. and it might be related to: #304

Huansion, 2021-05-10:
@drom Got it,that is a bug. 
 Wavedrom can draw beautiful waveform.Thank you for your work,sir.
The "svg open with Visio" problem is found when I want to modify the wrong wavedrom by Visio.The blank still exists when I copy the modified wavefrom to a word file.That's annoying.
And there is no extral blank when it stick to a word file directly ,but modify the wavefrom with word is inconvennient.
Hope you can solve these problems soon.Thank you very much.

drom, 2021-05-18:
How about this solution?

```
{signal: [
  {name: 'clk1', wave: 'l..N..'},
  {name: 'clk2', wave: 'l.hN..'},
  {name: 'clk3', wave: 'h..P..'},
  {name: 'clk4', wave: 'h.lP..'},
]}
```
[image]

`clk1` : Before the `Negative edge` you need to un-gate clock into `high` state.
`clk2` : that can be done with symbol `h` in the cycle right before the edge.

`clk3` : Before the `Positive edge` you need to un-gate clock into `low` state.
`clk4` : that can be done with symbol `l` in the cycle right before the edge.