wavedrom/wavedrom #323: SVG ids are not valid

wbrisett, 2021-04-26:
When exporting SVG images from Wavedrom, I discovered not all IDs are valid. According to the XML spec, IDs cannot begin with a numeric value, they must start with an alpha character. However, Wavedrom is using a combination of them. 

`    <defs>
        <g id="socket">
            <rect y="15" x="6" height="20" width="20"/>
        </g>
        <g id="pclk">
            <path d="M0,20 0,0 20,0" class="s1"/>
        </g>
        <g id="nclk">
            <path d="m0,0 0,20 20,0" class="s1"/>
        </g>
        <g id="000">
            <path d="m0,20 20,0" class="s1"/>
        </g>
        <g id="0m0">
            <path d="m0,20 3,0 3,-10 3,10 11,0" class="s1"/>
        </g>
        <g id="0m1">
            <path d="M0,20 3,20 9,0 20,0" class="s1"/>
        </g>`
        
       Here socket, pclk, and nclk are all valid, but all remaining values are not. Most tools seem to be allowing these, however, we have to validate all our content and images created with wavedrom are all failing due to this problem.

drom, 2021-05-06:
does it cause XML validation issues somewhere?

wbrisett, 2021-05-06:
It does we use epubcheck and when we have documents that contain wavedrom derived SVGs it will not pass those documents and thus we can't publish them. Looking at the raw SVG it really is a simple matter of following the standard XML ID rules and not using a numeric character.