wavedrom/wavedrom #324: [Q] Is there a way to put a "bar" over text to represent active low signals?

curiousengineer, 2021-05-06:
I want to write inverted signals. This would mean putting a horizontal line above the signal line. It is pronounced as "bar". I have not found a way to do this in wavedrom so far. Please let me know if this is possible at all. There is no mention of this in the tutorials.

curiousengineer, 2021-05-06:
This is what I tried but it did not work:

 {name: 'mosi', wave: 'x2222222222222220x2222222222222222x.', data: [ [{'text-decoration':'overline'}, 'a0'], 'a1', 'a2', 'a3', 'a4', 'a5', 'a6', 'a7', 'a8', 'a9', 'a10', 'a11', 'a12', 'a13', 'a14', 'd0', 'd1', 'd2', 'd3', 'd4', 'd5', 'd6', 'd7', 'd8', 'd9', 'd10', 'd11', 'd12', 'd13', 'd14', 'd15'], phase:0.5},

I tried to put 'a0 bar' into the wave.

drom, 2021-05-06:
You could use any decoration from this library: https://github.com/drom/tspan

Example:
```js
{signal: [
  {name: '<o>Q', wave: 'x3.4.5.x', data: ['<ins>Head', '<b>Tail</b>', '<s>tail'] }
]}
```

closed 2021-05-18 by drom