wavedrom/wavedrom #326: How to generate two clocks with different frequencies

tuuyii, 2021-06-07:
Hello, thanks for your work. 
but how to generate two clocks with different frequencies？

drom, 2021-06-07:
You can use `period` property on the wavelane like this:
```js
{signal: [
  {name: 'clk3', wave: 'p....', period: 3},
  {name: 'clk5', wave: 'p..',   period: 5},
]}
```
https://wavedrom.com/editor.html?%7Bsignal%3A%20%5B%0A%20%20%7Bname%3A%20%27clk3%27%2C%20wave%3A%20%27p....%27%2C%20period%3A%203%7D%2C%0A%20%20%7Bname%3A%20%27clk5%27%2C%20wave%3A%20%27p..%27%2C%20%20%20period%3A%205%7D%2C%0A%5D%7D

shygreyfield, 2021-06-28:
Could 'period' property be a decimal fraction?

heartacker, 2021-08-19:
> Could 'period' property be a decimal fraction?

not now, and i also want this.