wavedrom/wavedrom #329: Incorrect initialization sequence for 'U.0'

hackfin, 2021-06-15:
Using the [online editor](https://wavedrom.com/editor.html): 
```
{signal: [
  {name: 'clk', wave: 'p.....|...'},
  {name: 'gna', wave: 'U0..1.|..0'},
]}
```
the value of `gna` displays undefined up to the fourth clock cycle.

Inserting another 0:
```
{signal: [
  {name: 'clk', wave: 'p.....|...'},
  {name: 'gna', wave: 'U00.1.|..0'},
]}
```
displays as expected with the glitch at cycle 2.

Hope this is not a duplicate.

heartacker, 2021-08-19:
hi， you shouldn't use `U`, you need to use `u` in steal of `U` 。
and
you should use `.` when you have the same value. like`00`->`0.`
{signal: [
  {name: 'clk', wave: 'p.....|...'},
  {name: 'gna', wave: 'u0..1.|..0'},
]}

hackfin, 2021-08-19:
I'm aware of 'U' (caps) being not understood and taken as 'x', but the change to '0' should be reflected in the waveform.