wavedrom/wavedrom #335: how to draw sinusoidal wave using Wavedrom

13812003, 2021-09-02:
Hi everybody

Please help me to know how to draw sinusoidal waves with Wavedrom.

Thanks in advance

drom, 2021-09-26:
duplicate of https://github.com/wavedrom/wavedrom/issues/335

13812003, 2021-10-05:
Dear Drom, thanks for your response. How to do a ' duplicate of #335' is possible? Could you please let me know a few details?

smarshall54, 2022-12-21:
here is some code to write sines.
there is probably a way to encapsulate this into a function to make it easier to work with, as in the example of Step 9. of https://wavedrom.com/tutorial.html

```
{signal: [
  {name: 'clock',  wave: 'p..'},
  {name: 'sine',     wave: ['pw', {d: [
    'm',0,.5,
  	'q', .5,1,1,0,
    'q', .5,-1,1,0
]}]},
  {name: 'sine2',     wave: ['pw', {d: [
    'm',0,.5,
  	'q', 0.25,1,.5,0,
    'q', 0.25,-1,.5,0,
]}]},
    {name: 'cos',     wave: ['pw', {d: [
    'm',0,1,
    'q', 0.25,0,.5,-.5,
    'q', 0.25,-.5,.5,-.5,
    'q', 0.25,-.0,.5,.5,
    'q', 0.25,.5,.5,.5,
]}]},
  {name: 'cos2',     wave: ['pw', {d: [
    'm',0,1,
    'q', 0.125,0,.25,-.5,
    'q', 0.125,-.5,.25,-.5,
    'q', 0.125,-.0,.25,.5,
    'q', 0.125,.5,.25,.5,
  ]}]}
]}
```

heartacker, 2023-11-01:
hi @drom 
is it possible to add some new wave type name sin and cos and the user just need to set `freq`(base on tick) `amp` and `phase` 
[image]

```json
  [
    {
      "name": "sin_freq=1_phase=0_amp=1",
      "wave": [
        "sin",
        {
          "arg": {
            "f": 1,
            "phi": 0,
            "amp": 1
          }
        }
      ]
    },
    {
      "name": "cos_freq=1_phase=0_amp=1",
      "wave": [
        "cos",
        {
          "arg": {
            "f": 1,
            "phi": 0,
            "amp": 1
          }
        }
      ]
    },
    {
      "name": "cos_freq=1_phase=0_amp=1",
      "wave": [
        "sin", // use sin and set phase=45
        {
          "arg": {
            "f": 1,
            "ph": 45,
            "amp": 1
          }
        }
      ]
    },
  ]
```

heartacker, 2023-11-01:
and also add `repeat` 
repeat:0 (same the the largest ticks cycle)(**default**)
repeat:1 just one tick wave  (if freq =2, and there is 2 wave cycle in one tick)
repeat:-1 just one full sin/cos wave  (if freq=2 and then there is only 0.5 tick wave there)

Ganesh-AT, 2023-11-02:
The recently introduced piece-wise linear functionality is perfect for these kind of niche requirements. A proof-of-concept implementation is provided below for trigonometric functions with inputs as [specified](https://github.com/wavedrom/wavedrom/issues/335#issuecomment-1788436763). **It requires no changes to the core WaveDrom code**.

```

(function (res, splanes) {
	let sinTable = [];
	let cosTable = [];
	for (let i = 0; i < (2*3.1415926*1000); i++) {
		sinTable[i] = Math.sin(i/1000) ;
		cosTable[i] = Math.cos(i/1000) ;
	}	
	for (let l in splanes.trigLanes) {	
		let tlane = splanes.trigLanes[l] ;		
		if (typeof tlane.wave === 'undefined') {		
			res.signal.push(tlane);		
		} else {		
			let sigtmp = { name: tlane.name, wave : []};    		
			let tspec = tlane.wave ;		
			let dspec = tspec[1] ;
			let dspecamp = ((typeof dspec === 'undefined') || (typeof dspec.amp === 'undefined')) ? 1 : dspec.amp ;
			let dspecf = ((typeof dspec === 'undefined') || (typeof dspec.f === 'undefined')) ? 1 : dspec.f ;
			let dspecphi = ((typeof dspec === 'undefined') || (typeof dspec.phi === 'undefined')) ? 0 : dspec.phi ;
			let drepeat = ((typeof dspec === 'undefined') || (typeof dspec.repeat === 'undefined')) ? 1 : dspec.repeat ;		
			let pathArr = [] ;		
			for (let rn = 0; rn < drepeat; rn++) {
				for (let i = 0; i < (1/dspecf); i += 0.001) {
					let tblIdx = parseInt((2000 * 3.1415926 * i * dspecf) + (dspecphi * 1000 * 3.1415926 * dspecf / 180)) % parseInt(2000 * 3.1415926) ;
					let scaledVal = (tspec[0] === "sin") ? sinTable[tblIdx] : (tspec[0] === "cos") ? cosTable[tblIdx] : 0 ;
					// scaledVal is between -1 and +1, but pw coords pre-scaling should be between 0 and 1
					// first shift the value to be between -0.5 and 0.5 and then apply the amplitude factor
					scaledVal /= 2.00 ;
					scaledVal *= dspecamp ;
					// shift to center around y = 0.5
					scaledVal += 0.5 ;
					if ((rn == 0) && (i == 0)) {
						pathArr.push('M');
					}
					pathArr.push((rn / dspecf) + i);
					pathArr.push(scaledVal);
				}
			}		
			let pwObj = { 'd' : pathArr };      
			sigtmp.wave.push('pw', pwObj) ;  
	    	res.signal.push(sigtmp);			
		}		
	}
	return res ;	
})(
{ config: { skin:'default',hbounds:[0,30],hscale:1,marks:true,arcFontSize:8, arcStrokeSize: 1, fit2pane: 1 },
  head: {tick:0},
  signal: [    {name: 'CLK', wave: 'p.......................'},   ], },
{ trigLanes:   [
	{}, { "name": "sine",			"wave": [ "sin" ], },
	{}, { "name": "cosine", 		"wave": [ "cos" ], },
	{}, { "name": "s-f0.5-p0-a1",	"wave": [ "sin", {f: 0.5, phi: 0, amp:1} ], },
	{}, { "name": "s-f1-p0-a1",		"wave": [ "sin", {f: 1,   phi: 0, amp:1} ], },
	{}, { "name": "s-f2-p0-a1",		"wave": [ "sin", {f: 2} ], },
	{}, { "name": "s-f1-p0-a2",		"wave": [ "sin", {amp:2} ], },
	{}, { "name": "s-f1-p0-a0.5",	"wave": [ "sin", {amp:0.5} ], },
	{}, { "name": "c-f1-p0-a1",		"wave": [ "cos", {f: 1,   phi: 0,  amp:1} ], },
	{}, { "name": "c-f1-p45-a1",	"wave": [ "cos", {f: 1,   phi: 45, amp:1} ], },
	{}, { "name": "c-f0.5-p0-a3",	"wave": [ "sin", {f: 0.5, amp: 3} ], },
	{}, { "name": "sin_freq=1_phase=0_amp=1", "wave": [ "sin", { repeat : 5 } ] },
	{}, { "name": "sin_freq=1_phase=60_amp=1","wave": [ "sin", { phi: 60 } ] },
	{}, { "name": "s-f2-p0-a1-r12",		"wave": [ "sin", {f: 2, repeat: 12} ], },
	{}, { "name": "sine-r12",			"wave": [ "sin", {repeat: 12} ], },
	{}, { "name": "cosine-r12", 		"wave": [ "cos", {repeat: 12} ], },
	{}, { "name": "s-f0.5-p0-a1-r12",	"wave": [ "sin", {f: 0.5, phi: 0, amp:1, repeat : 12} ], },
	{}, { "name": "s-f1-p0-a1-r12",		"wave": [ "sin", {f: 1,   phi: 0, amp:1, repeat : 12} ], },
	{}, { "name": "s-f2-p0-a1-r12",		"wave": [ "sin", {f: 2, repeat : 12} ], },
	{}, { "name": "s-f1-p0-a2-r12",		"wave": [ "sin", {amp:2, repeat : 12} ], },
	{}, { "name": "s-f1-p0-a0.5-r12",	"wave": [ "sin", {amp:0.5, repeat : 12} ], },
	{}, { "name": "c-f1-p0-a1-r12",		"wave": [ "cos", {f: 1,   phi: 0,  amp:1, repeat : 12} ], },
	{}, { "name": "c-f1-p45-a1-r12",	"wave": [ "cos", {f: 1,   phi: 45, amp:1, repeat : 12} ], },
	{}, { "name": "c-f0.5-p0-a1-r12",	"wave": [ "sin", {f: 0.5, repeat : 12 } ], },
	{}, { "name": "sin_freq=1_phase=0_amp=1-r12", "wave": [ "sin", { repeat : 12 } ] },
	{}, { "name": "sin_freq=1_phase=60_amp=1-r12","wave": [ "sin", { phi: 60, repeat : 12 } ] },
    {},
  ]   
})

```

Output:

[image]

The 'repeat' functionality is not exactly as per the requirements specified in the follow-up comment, but it should be trivial to modify the current implementation to obey that.

heartacker, 2023-11-02:
great, wonderful process

drom, 2023-11-02:
here how sin can be done now:
```js
{name: 'sin2', wave: ['pw', {d: 'm,0,.5 q,.25,1,.5,0 t' + ',.5,0'.repeat(5)}]}
```
[image]

I think it is important to mix SIN/COS with other waveforms to show dynamic behavior. Like PLL Start:

```js
  {name: 'pll_en', wave: '0.1.....'},
  {name: 'pllout', wave: ['pw', {d: 'm,0,.5 h 4 q,.25,1,.5,0 t' + ',.5,0'.repeat(7)}]}
```
[image]

more experiments can be found here:
https://observablehq.com/@drom/piecewise?collection=@drom/wavedrom

heartacker, 2023-11-02:
thus，and we still need simple way for simple case

Ganesh-AT, 2023-11-02:
The PWL functionality in WaveDrom is incredibly flexible, but based on my own experience, it does not lead to 'readable' WaveJSON unless accompanied by extensive comments. The other features are much better in that aspect as the characters denoting different bricks are self-evident and 'self-documenting'. While 'M' and 'L' are easy to understand, quadratic Bezier curves are a challenge in keeping track of what each of the numbers in the 'q, x1, y1, x2, y2' mean. Even for other shapes, the path specifications for 'pw' are easier to develop if one is very familiar with the SVG grammar, but that might not be the case for everyone using WaveDrom. So, there is plenty of value in creating simple and understandable notations for like trigonometric functions like the one [requested](https://github.com/wavedrom/wavedrom/issues/335#issuecomment-1788436763). 

That said, I also see that much better value can be derived by placing these trigonometric waveforms along with other bricks in the same lane to show different aspects. Back in 2019, when I was tasked with creating waveforms to show aspects of the I2C protocol, DDR clocks, etc., I realized the importance of this feature (flexibility in combining different brick segments beyond the limitations imposed by WaveDrom itself) and developed an 'overlayOnLane' feature in my custom fork. My approach to creating PLL enable / lock, and PSK waveforms (with JS functions for new wave specs) in my custom fork is below (the WaveJSON can also be put into WaveDrom v3.3.0, but the output would be slightly different). Note that the JS function has been improved to accept just the original WaveJSON as input and doesn't need the trigonometric lanes to be separately specified. This allows inter-mingling of the trigonometric lanes with normal brick-based waveforms.

[image]

```

(function (res) {
  let sinTable = [];
  let cosTable = [];
  for (let i = 0; i < (2*3.1415926*1000); i++) {
    sinTable[i] = Math.sin(i/1000) ;
    cosTable[i] = Math.cos(i/1000) ;
  }
  for (let l in res.signal) {	
    let tlane = res.signal[l] ;
    if (typeof tlane.wave === 'object') {
      const triglane = (tlane.wave[0] == 'sin') || (tlane.wave[0] == 'cos');
      if (triglane) {
        let sigtmp = tlane ;
        let tphase = tlane.phase || 0 ;
        let tspec = tlane.wave ; // wave to mod
        tlane.wave = [] ;
        let dspec = tspec[1] ;
        let nodspec = (typeof dspec === 'undefined') ;
        let dspecamp = (nodspec || (typeof dspec.amp === 'undefined')) ? 1 : dspec.amp ;
        let dspecf   = (nodspec || (typeof dspec.f === 'undefined') || (dspec.f == 0)) ? 1 : dspec.f ;
        let dspecphi = (nodspec || (typeof dspec.phi === 'undefined')) ? 0 : dspec.phi ;
        let drepeat  = (nodspec || (typeof dspec.repeat === 'undefined')) ? 1 : dspec.repeat ;
        let pathArr = [] ;
        for (let rn = 0; rn < drepeat; rn++) {
          for (let i = 0; i < (1/dspecf); i += 0.001) {
            let tblIdx = parseInt((2000 * 3.1415926 * i * dspecf) + (dspecphi * 1000 * 3.1415926 * dspecf / 180))
                         % parseInt(2000 * 3.1415926) ;
            let scaledVal = (tspec[0] === "sin") ? sinTable[tblIdx] : (tspec[0] === "cos") ? cosTable[tblIdx] : 0 ;
            // scaledVal is between -1 and +1, but pw coords pre-scaling should be between 0 and 1
            // first shift the value to be between -0.5 and 0.5 and then apply the amplitude factor
            scaledVal /= 2.00 ;
            scaledVal *= dspecamp ;
            // shift to center around y = 0.5
            scaledVal += 0.5 ;
            if ((rn == 0) && (i == 0)) {
              pathArr.push('M');
            }
            pathArr.push((rn / dspecf) + i - tphase);
            pathArr.push(scaledVal);
          }
        }
        let pwObj = { 'd' : pathArr };
        tlane.wave = [ 'pw', pwObj ] ;
        res.signal[l] = tlane ;
      }
    }
  }
  return res ;	
})(
{ 
  config: { skin:'default',hbounds:[0,15.5],hscale:1,marks:true,arcFontSize:8, arcStrokeSize: 1, fit2pane: 1 },
  head: {tick:0},
  signal: [
    {name: 'clkp', wave: 'p..........'}, // Lane 0
    {},
    {name: "sinB", phase: 0, wave: [ 'sin', { f: 1 , repeat : 3, amp: 1 } ] }, // Lane 2
    {},
    {name: "sinC", phase: 0, wave: [ 'sin', { f: 0.5 , repeat : 3, amp: 1 } ] }, // Lane 4
    {},
    {name: 'pll_en',   wave: '0.1.........0...'}, // Lane 6
    {},
    {name: 'pll_out', wave: 'x..<.>'}, // Lane 8 needs 3 overlays - low freq sin, better freq sin, x
    {},
    {name: 'pll_lock', wave: '0......1.....0..'}, // Lane 10
    {},
    {name: 'data-digital', wave: 'h.l...h.l.'}, // Lane 12
    {},
    {name: "PSK-analog", phase: 0, wave: [ 'sin', { f: 1 , repeat : 2, amp: 1 } ] }, // Lane 14 needs 3 overlays
    {},
    {wave: [ 'tl', { coords: [1,0], text: '' } ], overlayOnLane: 14 }, // Lane 14 needs 3 overlays
    {phase: -3.5,    wave: [ 'sin', { f: 0.75 , repeat : 2, amp: 1 } ], overlayOnLane: 8}, // pll not locked
    {phase: -6.175,  wave: [ 'sin', { f: 1.25 , repeat : 9, amp: 1 } ], overlayOnLane: 8}, // pll locked
    {phase: -13.375, wave: 'x..', overlayOnLane: 8}, // pll not enabled
    {phase: -2, wave: [ 'sin', { f: 1 , repeat : 4, phi: 180 } ], overlayOnLane: 14 }, // data = h
    {phase: -6, wave: [ 'sin', { f: 1 , repeat : 2, phi: 0 } ], overlayOnLane: 14 }, // data = l
    {phase: -8, wave: [ 'sin', { f: 1 , repeat : 2, phi: 180 } ], overlayOnLane: 14 }, // data = h
  ], 
}
)

```

drom, 2023-11-02:
`overlayOnLane` does not look very intuitive to me.

drom, 2023-11-02:
> thus，and we still need simple way for simple case

```js
{signal: [
  {name: 'dummy', wave: '0...'},
  {name: 'analog',  wave: ['pw', {d: [
    'M',0,.5, // initial position
    ...(() => {
      const res = [];
      for (let i = 0; i < 65; i++) {
        res.push('L', i / 16, Math.sin(i * Math.PI / 8) / 2 + 0.5)
      }
      return res;
    })()
  ]
  }]},
]}
```
[image]

@heartacker is it simple enough?

Ganesh-AT, 2023-11-02:
> `overlayOnLane` does not look very intuitive to me.

When I started using WaveDrom heavily in 2019, I quickly got to a stage where I had to do plenty of manual editing of the generated SVGs. This was necessary for use-cases like creating delayed versions with dotted lines in the same lane, or DDR clocks with up and down arrows on both edges. We had a [discussion thread](https://github.com/wavedrom/wavedrom/issues/121#issuecomment-541269215) about this in late 2019. 

One of the most common edits I was doing was to take two different signal lanes and superimpose one on top of the other - just taking the argument to the transform / translate and changing the number to get the lane overlaid on another one. It was from this experience that I came up with the 'overlayOnLane' keyword. Obviously, if advanced waveforms are not being worked on, users don't need to take advantage of that (or even know that the feature exists).

I have seen many open issues that can be solved with the 'overlayOnLane' approach : #354 , #203, #201, #198 (to a certain extent), and #120, for example. As the comments in the discussion of the above issues show, different proposals were being examined for solving each of them. I may be partial to it, but the 'overlayOnLane' feature has been able to handle almost every curveball thrown at us in preparing professional datasheets and programming reference manuals.

[image]

drom, 2023-11-03:
@Ganesh-AT I think overlapping waveforms make sense in some cases.
I am not sure it is related to the topic of this particular issue.

heartacker, 2023-11-03:
> > thus，and we still need simple way for simple case
> 
> ```js
> {signal: [
>   {name: 'dummy', wave: '0...'},
>   {
>     name: 'analog',
>     wave: ['pw', {d: [
>       'M',0,.5, // initial position
>       ...Array.from({length: 64}).flatMap((_, i) =>
>         ['L', i / 16, Math.sin(i * Math.PI / 8) / 2 + 0.5])
>     ]}]
>   }
> ]}
> ```
> 
> [image]
> 
> @heartacker is it simple enough?

Thanks for the examples. It looks like we can be very creative with them, but as a non-technical user, I find Bézier curves to be quite challenging.

Your examples may be easy for you, but they're even more difficult for someone without programming experience.

I don't think I have the ability to understand your examples. They seem to require knowledge of JavaScript. I think most people don't understand them. If they don't understand them, they're even less likely to be able to write them.

I think things should be as simple as possible. In most cases, for simple scenarios, tools should be intuitive and focus on the needs, not the tool itself.

Of course, for complex scenarios, advanced methods can be used.

Thanks again.