wavedrom/wavedrom #353: wavedrom 2.9.1 install not working on linux

WilliamHeuser, 2022-03-01:
Hi 

I've tried to install wavedrom 2.9.1 on my machine running Rocky Linux 8 with the following steps:

-     Download the latest wavedrom-editor-v2.4.2-linux-{ia32|x64}.tar.gz release from here: [releases](https://github.com/wavedrom/wavedrom.github.io/releases)
-     unzip-untar the package: tar -xvzf wavedrom-editor-v2.3.2-linux-x64.tar.gz
-     Run the editor: ./WaveDromEditor/linux64/wavedrom-editor

But when I run the editor I get the following errors:

- [367268:367268:0301/091535.200362:FATAL:nw_content_renderer_hooks.cc(53)] Failed to load node library (error: libatomic.so.1: cannot open shared object file: No such file or directory)
- [0301/091535.234887:ERROR:file_io_posix.cc(144)] open /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq: No such file or directory (2)
-[367214:367230:0301/091535.539754:ERROR:chrome_browser_main_extra_parts_metrics.cc(230)] END: ReportBluetoothAvailability()

Thank you in advance for your help!

drom, 2022-03-07:
Does it start the editor?
What Linux, Version do you use?

WilliamHeuser, 2022-03-08:
Hi, 
Thank you for your reply!
No it never starts the editor. I'm opening it from the terminal and here it tries to open but it gets stuck and doesn't do anymore when reaching the errors: 
[367214:367230:0301/091535.539754:ERROR:chrome_browser_main_extra_parts_metrics.cc(230)] END: ReportBluetoothAvailability()
[414492:414500:0308/075542.818401:ERROR:cert_verify_proc_builtin.cc(603)] No net_fetcher for performing AIA chasing.

I'm running Rocky Linux release 8.5 (Green Obsidian)

WilliamHeuser, 2022-03-08:
[update]
I have found out that the problem was the error: 

[367268:367268:0301/091535.200362:FATAL:nw_content_renderer_hooks.cc(53)] Failed to load node library (error: libatomic.so.1: cannot open shared object file: No such file or directory)

Which could be resolved by running "dnf install libatomic" on my machine, so now it works locally.

As this is a module that could be loaded from a number of machines this is not a viable fix for me though. 
I'm not that knowledgeable about nw.js, but do you know if it is possible to package this library along with wavedrom.