wavedrom/wavedrom #364: Magic number: 166 -- where does this come from?

henrygab, 2022-06-21:
https://github.com/wavedrom/wavedrom/blob/8118f26283e6ecfd4304bdd58103c9621befc6d2/lib/append-save-as-dialog.js#L59

henrygab, 2022-06-21:
To see this corruption in action, as an example of how magic numbers cause fragility:

1. Open https://wavedrom.com/tutorial.html in favorite browser
2. Right-click on the first timing diagram (alfa)
3. Choose 'Save as SVG', and store it locally on your computer
4. Right-click on the second or later timing diagram (e.g., the clock diagram), 
5. Choose 'Save as SVG', and store it locally on your computer
6. Open both downloaded SVG files in your favorite editor

Expected results:
No truncation / loss of any aspect of the SVG contents.

Actual results:
Second (and later?) SVG data loses portions of the class name.  The following is *NOT* a typo:
`<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" id="svgcontent_1" height="300" width="400" viewBox="0 0 400 300" overflow="hidden" ass="WaveDrom">`

What happened:
Two characters are missing from 'class' (at 166 characters into the SVG text).  There is a strong connection between that character offset and the code that pulls in portions of the SVG data from index=0.

the first index is never affected, because although it splits the file at char 166, it concatenates it with the remainder of the `innerHtml` ... effectively making it a no-op for `index===0`.

Ganesh-AT, 2023-12-10:
This is an unfortunate vestige of not including the needed styles in all SVGs. Only the first SVG has that (related debug reported in #400 ). So, if the second SVG has to be saved standalone, the styles from the first one must be concatenated with the rest of the contents of the SVG to be saved. 

HTML specs allow the styles to leak across all SVGs if specified anywhere in the document, and WaveDrom has taken 'advantage' of that. One can prepare a band-aid PR that tries to delve deeper into the SVG structure to be saved and then make the appropriate assembly. This can avoid the 166 magic number.

However, the long-term reliable solution would be to allow styles to be embedded within each SVG, and prevent leakage by wrapping SVGs in a IMG tag while embedding. A chicken bit 'wrapSvgInImg' can be made available in the config for direct rendering (useful for debug / developers), but it should be true by default to allow seamless assembling of multiple SVGs on the same page.