wavedrom/wavedrom #369: PNG background options

jimga150, 2022-07-28:
I am using the web version.

When exporting to a PNG, it looks like It has a transparent background, which makes the waveforms illegible if a dark background is used in whatever image viewer someone is using. Can the PNG be modified (or given the option) to use a solid white background?

With white background:
[image]

With black background, such as in the default windows photo app:
[image]

rgetz, 2022-12-22:
+1 for handling dark mode applications

grant-h, 2023-04-07:
For many, the default on GitHub is dark mode, which makes renderings illegible.

closed 2023-07-02 by drom

CAESIUS-TIM, 2023-10-16:
- vscode
  - version: 1.83.1
  - color theme: Dark Modern (Default Dark Modern)
- Markdown Preview Ext.: markdown-all-in-one 
  - version: v3.5.1
  - preview theme: None
  - [Repository](https://github.com/yzhang-gh/vscode-markdown)
  - [Marketplace](https://marketplace.visualstudio.com/items?itemName=yzhang.markdown-all-in-one)

```markdown
\`\`\`wavedrom
{ 
  signal: [
    { name: "SIO_C", wave: '01.0.1.0.1.0|1.'},
    { name: "SIO_D (IN)", wave: '1.0.=...=...|010', phase: 0.5 },
    { name: "SIO_D (OUT)", wave: 'x...=...=...|...', phase: 0.5 },
  ],
  config: {
    hscale: 1,
  },
}
\`\`\`
```
<s>How to escape the backtick symbol in GitHub Markdown? Different from testing in vscode.</s>

[image]

# Dark skin

But no stable or merged.

#366
#114

commit 5f5ebc22 (2023-07-02): fixes #369 ; brutal white backdrop under every waveform