wavedrom/wavedrom #372: Glitches after Spacer

wtl0, 2022-09-19:
Thank you for this perfect tool, I've been looking for such since long. However, after using sporadically (online editor), I've found an issue.

Issue: After a spacer, there might show up a glitch in the rendering. Please compare picture below, red cycle.
[image]

The same glitch occurs, when the spacer is moved to an earlier position.
`  {name: 'ack', wave: '1...|..10.'},`
[image]

wtl0, 2022-09-19:
Maybe this is a feature instead of a bug, the same occurs without spacers:
`{ "signal" : [
{ "name": "fine", "wave": "1..0.." },
{ "name": "documentation gap?",  "wave": "1.100." },
]}`
[image]
Obviously  `"00"` is not the same as `"0."`
?

DmitTrix, 2023-02-19:
I would treat this as a feature rather than a bug.  It's useful to represent glitches (as a [real electrical issue](https://en.wikipedia.org/wiki/Glitch#Electronics_glitch), not just a rendering problem) when needed.

Proper visual rendering can be achieved in all cases by using the `.` ("repeat previous value") command instead of `0` or `1`.

Formally documenting this would be good, though.

drom, 2023-03-10:
Documentation PR is welcome