wavedrom/wavedrom #375: path not handled to the fullest extent...

rgetz, 2022-12-28:
Sorry in advance for the length - it's just good to have all the info in one spot to discuss.

With the recent [add](https://github.com/wavedrom/wavedrom/commit/7c35f8017efdca1a560df86d312f9bfc4e7173ac) of piecewise waves - is a great way for users to add analog waves to their timing diagrams, and should close a few issues (#68, #189, #317, and partially #198).

Now users can support things like this:
```
{signal: [
  {name: 'clock',    wave: 'p.........'},
  {name: 'A', wave: '01'.repeat(5)},
  {name: 'sin(a)', wave: ['pw', {d: 'm,0,.5 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0 q,.25,1,.5,0 q,.25,-1,.5,0'}]},
  {name: 'sin(b)', wave: ['pw', {d: 'm,0,.5 q,.25,1,.5,0 t,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0'}]},
]}
```

However, there are svg options which were not implemented in this PR, which would make the actual JSON a little shorter, and cleaner. (in the above example - `sin(a)` and `sin(b)` will render to the same thing (after the issue below are implemented, which I have a PR for).

based on the existing PR, and [svg spec/doc](w3.org/TR/SVG/paths.html) WaveDrom now supports:

**`moveto` command**

| command | name | Parameter | Description |
|----------------|----------|-----------------|-------------------|
| `M` (absolute), `m` (relative) | moveto | (x y)+ |  Start a new sub-path at the given (x,y) coordinates. M (uppercase) indicates that absolute coordinates will follow; m (lowercase) indicates that relative coordinates will follow. If a moveto is followed by multiple pairs of coordinates, the subsequent pairs are treated as implicit lineto commands. Hence, implicit lineto commands will be relative if the moveto is relative, and absolute if the moveto is absolute. If a relative moveto (m) appears as the first element of the path, then it is treated as a pair of absolute coordinates. In this case, subsequent pairs of coordinates are treated as relative even though the initial moveto is interpreted as an absolute moveto.|

However, WaveDrom doesn't support that exactly, it only supports `(x y)` coordinates, not `(x y)+` coordinates (_multiple pairs of coordinates_). For the `moveto` command - I agree it doesn't make a lot of sense to add this (moving around, and not drawing anything seems sort of pointless, but it is is more of an issue for the other drawing commands...

**`lineto` commands**

| command | name | Parameter | Description |
|----------------|----------|-----------------|-------------------|
| `L` (absolute) `l` (relative) | lineto | (x y)+ | Draw a line from the current point to the given (x,y) coordinate which becomes the new current point. L (uppercase) indicates that absolute coordinates will follow; l (lowercase) indicates that relative coordinates will follow. A number of coordinates pairs may be specified to draw a polyline. At the end of the command, the new current point is set to the final set of coordinates provided.|
|`H` (absolute) `h` (relative) | horizontal lineto | x+ | Draws a horizontal line from the current point. H (uppercase) indicates that absolute coordinates will follow; h (lowercase) indicates that relative coordinates will follow. Multiple x values can be provided (although usually this doesn't make sense). An H or h command is equivalent to an L or l command with 0 specified for the y coordinate. At the end of the command, the new current point is taken from the final coordinate value. |
| `V` (absolute) `v` (relative) | vertical lineto | y+ | Draws a vertical line from the current point. V (uppercase) indicates that absolute coordinates will follow; v (lowercase) indicates that relative coordinates will follow. Multiple y values can be provided (although usually this doesn't make sense). A V or v command is equivalent to an L or l command with 0 specified for the x coordinate. At the end of the command, the new current point is taken from the final coordinate value. |

Same issue, WaveDrom does not support the polyline shorthand of _multiple coordinates_ following the initial ones.

**elliptical arc curve commands**

| command | name | Parameter | Description |
|----------------|----------|-----------------|-------------------|
| `A` (absolute) `a` (relative) | elliptical arc | (rx ry x-axis-rotation large-arc-flag sweep-flag x y)+ | Draws an elliptical arc from the current point to (x, y). The size and orientation of the ellipse are defined by two radii (rx, ry) and an x-axis-rotation, which indicates how the ellipse as a whole is rotated, in degrees, relative to the current coordinate system. The center (cx, cy) of the ellipse is calculated automatically to satisfy the constraints imposed by the other parameters. large-arc-flag and sweep-flag contribute to the automatic calculations and help determine how the arc is drawn. |

Same issue, WaveDrom does not support the polycurve shorthand.

**quadratic Bézier curve commands**

| command | name | Parameter | Description |
|----------------|----------|-----------------|-------------------|
| `Q` (absolute) `q` (relative) | quadratic Bézier curveto | (x1 y1 x y)+ | Draws a quadratic Bézier curve from the current point to (x,y) using (x1,y1) as the control point. Q (uppercase) indicates that absolute coordinates will follow; q (lowercase) indicates that relative coordinates will follow. Multiple sets of coordinates may be specified to draw a polybézier. At the end of the command, the new current point becomes the final (x,y) coordinate pair used in the polybézier. |
| `T` (absolute) `t` (relative) | Shorthand/smooth quadratic Bézier curveto | (x y)+ | Draws a quadratic Bézier curve from the current point to (x,y). The control point is assumed to be the reflection of the control point on the previous command relative to the current point. (If there is no previous command or if the previous command was not a Q, q, T or t, assume the control point is coincident with the current point.) T (uppercase) indicates that absolute coordinates will follow; t (lowercase) indicates that relative coordinates will follow. At the end of the command, the new current point becomes the final (x,y) coordinate pair used in the polybézier.|

This time - WaveDrom just doesn't support the `T` or `t` commands at all, and doesn't support the polybézier shorthand.

**Conclusion**

add support for:
  * the `t` and `T` `Shorthand/smooth quadratic Bézier curveto` command, 
  * polybézier & polylines; 
which is minimally invasive in `lib/render-piece-wise.js`.

Comments welcome.

drom, 2023-01-06:
Great job @rgetz !

rgetz, 2023-01-06:
Thanks for merging - closing this now.

closed 2023-01-06 by rgetz