wavedrom/wavedrom #377: Different Threshold Levels

alpCaner, 2023-01-03:
Hi,

We would like to have the following waveform which has different threshold levels. In wavedrom currently there is only 50% threshold by default. 

[image]

How can we achieve it?

Regards,
Caner

drom, 2023-01-05:
@alpCaner interesting idea.
What kind of syntax would you imagine to describe multiple thresholds?
Are you thinking of 10% and 90% levels only, or arbitrary level?

alpCaner, 2023-01-05:
For the syntax wise I'm not sure. TH10-90 maybe to describe falling and rising edge.

For this case it is 10% and 90% levels. Sometimes we have 20% and 80% levels, but in this small scale it wouldn't matter. But making it arbitrary is more helpful I believe. It can be very flexible.