wavedrom/wavedrom #378: Tick and Tock Broken

DrMatthewJApplegate, 2023-01-13:
https://wavedrom.com/editor.html

Tick and tock no longer work. Here's a testcase.

{signal: [
    {name: 'clock', wave: 'P...........'},
  ],
  head:{
   text:'WaveDrom example',
   tock:1,
   every:1
 }
}

gmoreto, 2023-04-19:
it still works but it seems 'every' needs to be changed to 'per'

Ganesh-AT, 2023-04-29:
Problem crept in due to the update from var to let for render-marks.js

Line 34 declares _offset_ as const in local scope of the code segment that handles tick / tock value being a single number, but _offset_ is a 'let' variable that is supposed to be available everywhere in the function.

Line 34 of https://github.com/wavedrom/wavedrom/blob/trunk/lib/render-marks.js

cc @drom

commit b3e324dd (2023-07-02): fixes #378
closed 2023-07-02 by drom