wavedrom/wavedrom #379: Common gap lines

drom, 2023-01-22:
Working on new proposal for common gap lines.

Composite gap lines can be useful to describe grouping, conditionality and repetition withing timing digaram?

[image]

sam-falvo, 2023-01-23:
Looks OK to me; I'm not sure how the syntax of the tags member correlates with the resulting picture though.  The spaces and dots don't seem to align with any visual feature.  The (, ), |, and S are fairly self-evident.

drom, 2023-01-23:
> Looks OK to me; I'm not sure how the syntax of the tags member correlates with the resulting picture though. The spaces and dots don't seem to align with any visual feature. The (, ), |, and S are fairly self-evident.

These are space separated (Forth style) words ;)

rgetz, 2023-02-08:
This looks pretty interesting.

How does it work with a piecewise wave in the middle?
`{name: 'sin(b)', wave: ['pw', {d: 'm,0,.5 q,.25,1,.5,0 t,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0,.5,0'}]},`

Other question would be if you can change the location or clock phase of where the lines are.