wavedrom/wavedrom #384: Can't access svg.wavedrom.com

BrewKris, 2023-03-29:
I'm trying to use the svg.wavedrom.com to render an image in a markdown document.

I've followed the instructions in the README.md, but I can't get the image to appear as it does in the README.md

If I try to go directly to svg.wavedrom.com I get {"message":"Missing Authentication Token"}

Is there something I need to do to use the server?

BrewKris, 2023-03-29:
My repository is on bitbucket cloud. I did a quick test from a github repository and the image does render ok.

drom, 2023-07-02:
What is the exact URL you are sending?

BrewKris, 2023-07-02:
My example files are at https://github.com/BrewKris/testWavedrom#readme

Example address is https://svg.wavedrom.com/github/BrewKris/testWavedrom/main/status.json5

This works fine when the source .json5 is within github. If the file is in a bitbucket.org repository then it will not display, I just get the error message from the server.

LiveQiu, 2023-09-16:
I'm having the same issue as you, I can't call this URL to generate a timing graph in my local preview，I don't know if there is any other way to solve it

drom, 2023-09-16:
it only work on github
if you want bitbucket, please file the issue here https://github.com/wavedrom/lambda