wavedrom/wavedrom #409: Dark Theme Renders on White Background

reedbn, 2024-04-02:
In Issue #369 rendering a theme expecting a light background on a dark background caused issues. Now, we have the opposite issue, where rendering the dark theme will be put on a white background, making it unusable in the opposite direction.

For example, here's a common test of `{ signal: [{ name: "Alfa", wave: "01.zx=ud.23.4567890" }], config:{skin:"dark"}}` (slightly modified with the config on the end), as saved out in PNG by the [the editor](https://wavedrom.com/editor.html):
[image]

We either need a new mechanism that allows each theme to specify the preferred background color, or alternatively provide a method of explicitly configuring this option.

whitequark, 2024-04-04:
Take a look [here](https://github.com/YoWASP/wavedrom/blob/main/lib/api.js), I've added proper browser color scheme support with media queries.

reedbn, 2024-04-12:
@whitequark using media query selectors is a great idea! Are you planning on making a PR for your changes? I made a very hacky fix for this with PR #410, but would love to see something more robust land instead, if available.

whitequark, 2024-04-12:
I'm OK with my solution staying within YoWASP, as I found working with this codebase to be excessively frustrating and time-consuming.