wavedrom/wavedrom #430: Create a tutorial for visualizing data from a logic analyzer

hackerb9, 2025-04-11:
The tutorials currently only show how to generate artificial signals and schematics. But, the only references I can find to WaveDROM being used with actual data from a logic analyzer are to "BeagleLogic," which appears to be a defunct DIY project. Unfortunately, it does not even mention WaveDROM in its [documentation](https://beaglelogic.readthedocs.io/en/latest/).

From the little I understand so far, it seems I'll need to use some tool for converting raw data from a logic analyzer into WaveJSON. I would love to see a short tutorial showing either how to use the tool included in WaveDROM (if such exists) or how to write one's own (if it doesn't).  Or, I suppose, as a third possibility, it might be helpful to point folks to alternate projects if WaveDROM is not appropriate for visualizing measured data. Thanks!

drom, 2025-07-10:
WaveJSON is not meant for large traces. You can use VCD instead.

hackerb9, 2025-07-10:
Thanks for the suggestion to use the VCD file format. It looks like a lot of logic analyzers save directly to VCD, so I'm guessing you are saying WaveDrom is not the program I should be using.

If that's right, please consider adding the following to WaveDrom's documentation:

> **NB**: WaveDrom is for handmade timing diagrams and is not appropriate for real-world data due to the size limitations of the WaveJSON format. 
>
> For viewing data measured from a logic analyzer, consider one of these fine programs instead:. 
> 1. [VCDrom](https://github.com/wavedrom/vcdrom) 
> 2. 
> 3.

drom, 2025-07-10:
Yes, that is exactly a reason we implementing VCDrom.

hackerb9, 2025-07-10:
Thanks! Then my request for a tutorial is moot and I will close this issue. Feel free to reopen if you want the reminder that the README.md does not yet mention VCDRom.